ondryaso/pi-rc522 #6: does not work with Raspian Jessie?

Sispheor, 2016-08-06:
It seems something changes ioctl message structure.
See https://wolfgangklenk.wordpress.com/2015/10/25/spi-communication-issues-with-raspbian-jessie/

faultylee, 2016-08-08:
Looks like an upstream issue between kernel (raspberrypi/linux#1547) and SPI-py (https://github.com/lthiery/SPI-Py/issues/17)

@Sispheor doesn't look like it's related to ioctl and the article you linked was from Oct 2015. I had this working for me in Apr 2016 on `Jessie`.

Sispheor, 2016-08-08:
> lthiery/SPI-Py#17 <https://github.com/lthiery/SPI-Py/issues/17>)

Yes, the first article is outdated. And Yes it was actualy an issue in
SPI-py.
Thx for your answer.

ondryaso, 2016-08-09:
So it's needed to use custom SPI-py with the cs_change set to 0, do I understand this correctly?

Sispheor, 2016-08-09:
Yes !

ondryaso, 2016-08-09:
Alright, I'm gonna mention this in the README and close this.

closed 2016-08-10 by ondryaso