ondryaso/pi-rc522 #8: Integration into other project

PierreRust, 2016-08-31:
I find it difficult to integrate this lib into other python project : as it has not setup.py to install it,  it must be included directly into the source of the application using it.
That's not really a problem, except that RFID.py and RFIDUtil.py import each other, meaning that if you simply copy the ChipReader directory in the application source tree, these import fails (there are not relative import).

AFAIK there are several way of solving this kind of problem, the cleanest it probably to make it an actual library (with a setup.py). I could work on a pull request for this if you want.

faultylee, 2016-09-01:
I guess it's no harm, though I personally like "xcopy" install, no lib dependency. 

IIRC, I place my code alongside `RFID.py` and `import RFID` works. Take a look at `Read.py`

PierreRust, 2016-09-01:
I agree, for a simple lib, a "copy" install is generally fine and convenient. But I like to avoid putting the lib files in the middle of the other files of my application, so I generally put them in a subdirectory.

The problem here is the way  RFID.py and RFIDUtil.py import each other. Maybe a very simple solution would simply be to all the code in a single and avoid these imports altogether. Is there any specific reason for having two separate files ?

BTW, thanks for the lib !

faultylee, 2016-09-01:
I didn't really use `RFIDUtil.py`, if not mistaken it's to group helper function while `RFID.py` is actual RFID functions. I guess it's due to the concept of separation of concern.

I also a happy user :+1:

ondryaso, 2016-09-02:
Actually, I might have a look on PyPI and setup methods, I haven't really been studying this before.

thijstriemstra, 2016-09-09:
I'll make a pull request. @ondryaso you should register an account on https://pypi.python.org so `pip install pi-rc522` can be created in the future.

thijstriemstra, 2016-09-09:
@ondryaso created pull request #13, please take a look.

ondryaso, 2016-09-10:
Thanks @thijstriemstra 
https://pypi.python.org/pypi/pi-rc522/1.1.0

closed 2016-09-10 by ondryaso