ondryaso/pi-rc522 #11: Can't read HID iClass

thwillingham, 2016-09-03:
I cannot get any HID iClass DH cards to work.  I'm not sure if it's a software or hardware issue.  Any help is much appreciated.

faultylee, 2016-09-03:
HID iClass cards is using [ISO 15693 standard](
http://www.secureidnews.com/news-item/the-different-contactless-smart-card-flavors-many-types-but-do-they-play-together/) which is different from ISO 14443/A which [RC522 support](http://www.nxp.com/documents/data_sheet/MFRC522.pdf)

thwillingham, 2016-09-03:
Do you know of a similar reader that supports ISO 15693?

faultylee, 2016-09-03:
I would usually look for a reader that works with Arduino, and then try to see if someone ported the code to Pi

http://forum.arduino.cc/index.php?topic=298490.msg2172778#msg2172778

closed 2016-09-03 by thwillingham