ondryaso/pi-rc522 #12: Two Readers (CS0 and CS1) not working in one program

Thmas47051, 2016-09-08:
Hi,
using one reader (spidev0.0 and spidev0.1) in one program works (Read.py).

Two readers are not working. Only the first initialized is working.
Any idea?

Thomas

`import RFID
import signal
import time

run = True
rdr0 = RFID.RFID("/dev/spidev0.1")
rdr1 = RFID.RFID("/dev/spidev0.0")
util1 = rdr1.util()
util1.debug = True
util0 = rdr0.util()
util0.debug = True

def end_read(signal,frame):
    global run
    print ("\nCtrl+C captured, ending read.")
    run = False
    rdr0.cleanup()
    rdr1.cleanup()

signal.signal(signal.SIGINT, end_read)

print ("Starting")
while run:
    (error, data) = rdr0.request()

    (error, uid) = rdr0.anticoll()
    if not error:
        print ("In Card read UID: "+str(uid[0])+","+str(uid[1])+","+str(uid[2])+","+str(uid[3]))
       
    (error, data) = rdr1.request()
    (error, uid) = rdr1.anticoll()
    if not error:
        print ("Out Card read UID: "+str(uid[0])+","+str(uid[1])+","+str(uid[2])+","+str(uid[3]))

     time.sleep(0.1)
`

Thmas47051, 2016-09-17:
Solved.
The trick was:
rdr2 = RFID.RFID("/dev/spidev0.0",pin_ce=24)
rdr1 = RFID.RFID("/dev/spidev0.1",pin_ce=26)

Bye,
Thomas
`
import RFID
import signal
import time

run = True

rdr2 = RFID.RFID("/dev/spidev0.0",pin_ce=24)
rdr1 = RFID.RFID("/dev/spidev0.1",pin_ce=26)

def end_read(signal,frame):
    global run
    print ("\nCtrl+C captured, ending read.")
    run = False
    rdr1.cleanup()
    rdr2.cleanup()

signal.signal(signal.SIGINT, end_read)

print ("\nStarting")
while run:
    print("\n******")
    (error, data) = rdr1.request()
    if not error:
        print ("\nDetected: " + format(data, "02x"))

        (error, uid) = rdr1.anticoll()
        if not error:
          print ("Card read UID IN : "+str(uid[0])+","+str(uid[1])+","+str(uid[2])+","+str(uid[3]))
    (error, data) = rdr2.request()
    if not error:
        print ("\nDetected: " + format(data, "02x"))
        (error, uid) = rdr2.anticoll()
        if not error:
          print ("Card read UID OUT : "+str(uid[0])+","+str(uid[1])+","+str(uid[2])+","+str(uid[3]))
`

EVENFATE, 2016-09-21:
still not working ,for help,readers cant read the tag data

EVENFATE, 2016-09-21:
i use gpio expansion board to set up two readers in one raspberry ,but the program not work

EVENFATE, 2016-09-21:
please help ....

erivandoramos, 2016-11-19:
@EVENFATE 
I present a solution here: [](https://github.com/erivandoramos/TwoRC522RPi)

LudwigKnuepfer, 2017-01-03:
@ondryaso I guess this issue can be closed.

PsuFan, 2019-02-27:
You forgot the rst pin.

rdr1 = RFID.RFID(device=0, pin_ce=24, pin_rst=22)
rdr2 = RFID.RFID(device=1, pin_ce=26, pin_rst=18)