ondryaso/pi-rc522 #15: RFID-THM3030

thijstriemstra, 2016-10-05:
Do you have any idea if and how this [reader](http://www.ebay.com/itm/RFID-Reader-Module-Non-Touch-Board-Compatible-RC522-Module-UART-SPI-Interface-/331949515040?_trksid=p2047675.l2557&ssPageName=STRK%3AMEBIDX%3AIT&nma=true&si=y6BjiV9QZ2eTyLOyd0ZFe3XsJzI%253D&orig_cvip=true&rt=nc) pictured below would map to the pins mentioned in the documentation?

[image]

faultylee, 2016-10-06:
That looks like any RC522 module, try

| Board pin name | Board pin | Physical RPi pin | RPi pin name | Your Module  |
|----------------|-----------|------------------|--------------|--------------|
| SDA            | 1         | 24               | GPIO8, CE0   | SS_N         |
| SCK            | 2         | 23               | GPIO11, SCKL | SCLK         |
| MOSI           | 3         | 19               | GPIO10, MOSI | MOSI         |
| MISO           | 4         | 21               | GPIO9, MISO  | MISO         |
| GND            | 6         | 6, 9, 20, 25     | Ground       | GND          |
| RST            | 7         | 22               | GPIO25       | RSTN         |
| 3.3V           | 8         | 1                | 3V3          | DVDD         |

thijstriemstra, 2016-10-12:
Thanks for your feedback @faultylee, I'll give that a try.

ondryaso, 2016-10-29:
@thijstriemstra Well, does it work? I have trouble finding any info about it.

thijstriemstra, 2016-10-29:
ill give it a try this weekend, thanks for your patience.

eboily, 2016-11-22:
Any news? Does it work? Any insights to share? :-)

thijstriemstra, 2016-11-22:
I connected it using @faultylee's instructions but no response when I swipe over it. It also seems it doesn't have a LED indicator so it's hard to say what it's doing.