ondryaso/pi-rc522 #24: Requesting serial(UART) RC522 for Arduino

cyberjuputer, 2017-02-02:
Hi there.

I see that you only code for Python. I hope you code one for Arduino also.

Thank you.

faultylee, 2017-02-03:
@cyberjuputer There's already existing library for Arduino using SPI. See
http://playground.arduino.cc/Learning/MFRC522
https://github.com/miguelbalboa/rfid

cyberjuputer, 2017-02-03:
@faultylee Hi there. I know that and I have used that one before. However I need the one that uses Serial communication instead of SPI. I clearly do not know how to write the library myself.

faultylee, 2017-02-03:
I think it makes more sense to ask from miguelbalboa rather than here, since you're looking for the same RC522 functionality for Arduino but via serial. As far as I understand the protocol is the same, just need to send the same byte stream via serial instead of SPI. Porting python to Arduino requires duplicating unnecessary code already exist in  https://github.com/miguelbalboa/rfid

closed 2017-02-23 by ondryaso