ondryaso/pi-rc522 #29: Can't read cards

arough, 2017-05-28:
I changed from MFRC522-python because of the "list index out of bounds" issue that has been mentioned there.

If I run the Read.py and swipe my card nothing happens. No error, no nothing.
I'm on a Pi 2B, raspbian, everything is wired correctly and I installed spidev.
If I type "ls /dev/spi*" there is spidev0.0 and spidev0.1. Anything wrong there?

Everything was setup correctly for my project on MFRC522-python and now I have to implement it with pi-rc522 and it stops me right at the beginning :'(

arough, 2017-05-28:
Well I just fixed the issue with MFRC522-python so the problem won't affect me anymore ;)

If you'd like you can close this issue or keep it open for someone with the same problem.

closed 2017-07-05 by ondryaso

mattelacchiato, 2018-11-04:
I'm having the same problem. I'm coming from "MFRC522-python", too. Using the `Read.py` from examples, I get this output:

```Starting

Detected: 10
Card read UID: 114,32,84,195
Setting tag
Selecting UID [114, 32, 84, 195, 197]

Authorizing
Changing used auth key to [116, 0, 82, 53, 0, 255] using method B

Reading
Calling card_auth on UID [114, 32, 84, 195, 197]
Error on S1.0B0

Deauthorizing
Changing auth key and method to None
```
Can anyone point me in a direction how to fix this? Going back to "MFRC522-python" is not an option for me because of python2...

mattelacchiato, 2018-11-04:
I've resolved this by changing the auth to this line:
```util.auth(rdr.auth_a, [0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF])```

tslater, 2022-12-16:
I'm having the same error with NFC215 tags. @mattelacchiato's fix didn't work for me. Has anyone gotten those to work? Or can anyone tell me if I need a new auth key? Not sure what to do there.