ondryaso/pi-rc522 #30: Reading card content to text

o15a3d4l11s2, 2017-06-25:
Could you please include or point me to an example for how to read the whole content of a card and printing it as a string?

I have checked your examples but could not wrap my head around the whole logic for reading the data from the card.

I understand the sectors and blocks, but what exactly should be done in order to read the whole card and output the whole data as a textual content?

mattelacchiato, 2018-11-04:
Got this working. But keep in mind that this lib is heavily byte-oriented. I'll guess, you'll have to figure out the corresponding formats for your own...

This is how i read out the first 20 blocks:

```python
def read_out(block_address):
    if not util.is_tag_set_auth():
        return True

    error = util.do_auth(block_address)
    if not error:
        (error, data) = util.rfid.read(block_address)
        #print(util.sector_string(block_address) + ": " + data)
        print(util.sector_string(block_address) + ": " + "".join(chr(e) for e in data))
    else:
        print("Error on " + util.sector_string(block_address))

for i in range(0,19):
    read_out(i)
```

o15a3d4l11s2, 2018-11-05:
Thanks a lot, @mattelacchiato, will try this out in the next days.

ogroef, 2018-11-22:
> Got this working. But keep in mind that this lib is heavily byte-oriented. I'll guess, you'll have to figure out the corresponding formats for your own...
>
> This is how i read out the first 20 [...]

How did you, @mattelacchiato, handle invalid unicode values.

Kind
Olivier

mattelacchiato, 2018-11-22:
Yeah, this lib is indeed very byte-oriented. I've found this article very helpful: https://learn.adafruit.com/adafruit-pn532-rfid-nfc/mifare

I don't read the content as string, so I don't have a problem with wrong unicode values. I read only NDEF format (https://learn.adafruit.com/adafruit-pn532-rfid-nfc/ndef) and pass those bytes to a NDEF lib (https://ndeflib.readthedocs.io/en/stable/). But which bytes to read and how many can be tricky, too ;-)

Woppilif, 2019-04-27:
> Yeah, this lib is indeed very byte-oriented. I've found this article very helpful: https://learn.adafruit.com/adafruit-pn532-rfid-nfc/mifare
>
> I don't read the content as string, so I don't have a [...]

Can you please show your snippet of this code?

mattelacchiato, 2019-04-29:
Sure: https://github.com/mattelacchiato/hoerspieler/blob/master/main.py#L44

Hope this helps...

Woppilif, 2019-04-29:
> Sure: https://github.com/mattelacchiato/hoerspieler/blob/master/main.py#L44
> 
> Hope this helps...

Thank you! 
Have you tried to read tags from phone?
I'm getting some errors.

mattelacchiato, 2019-04-29:
Jeah, that works in my case. You have to be compliant to the standard...

Woppilif, 2019-04-29:
> […](#)

Tell me please which app do you use to send data between devices.
And is this complete code that you use to do this? https://github.com/mattelacchiato/hoerspieler/blob/master/main.py#L44
Have you changed rc522 lib?
I'm trying to understand this case but still geting nothing.

mattelacchiato, 2019-04-29:
I use this only for communication between the NFC module and an NFC card,
not between devices.

This is my complete code, i didn't changed anything. But I've copied and
modified some methods from pi-rc522 to fit them for my needs.

Woppilif, 2019-04-29:
> I use this only for communication between the NFC module and an NFC card, not between devices. This is my complete code, i didn't changed anything. But I've copied and modified some methods from [...]

Ah. Okay thou. Now it's become absolutely clear for me. As I see there is no opportunity to use rc522 and NFC on Phone.

Thank you btw!