ondryaso/pi-rc522 #40: Read faster?

Ogg3l, 2018-03-07:
Hi,

first of all thank you for you rc bib. It works fine :)
Do you have any idea to read faster? We have the RFID reader under remote-controlled cars and the Tags lie on the ground. The problem is, that the cars drive too fast over the tags, so that the reader can't read.

We have no idea to improve that. Maybe you have an idea? :)

Thank you :)

Divyank95, 2018-03-08:
This problem may be solved by increase the range of rfid reader by using
ACTIVE RFID TAGS that lies on ground.

Ogg3l, 2018-03-08:
Ahh okay thank you for the info. We thought that there is maybe an option to improve the read rate or something like that. Is there an option we could test without Active Tags? Do you have an idea?

Divyank95, 2018-03-08:
since mfrc522 works on spi protocol, you can increase the data transfer
rate by increasing the serial clock sck of master device.

And if you do not want to use active tags then you can also connect a long
range antenna, which is able to transmit more power, to mfrc522 module.

Ogg3l, 2018-03-08:
Can you explain how to increase the clock rate? We are newbies on this issue.

closed 2018-04-12 by Ogg3l