ondryaso/pi-rc522 #48: Read fails after some time

zlokomatic, 2018-11-25:
Hi,

i got two problems.

1. wait_for_tag is not working for me - it just hangs there waiting. However i can read tags after commenting out the wait_for_tag bit
2. After like 40 reads in a row ( not a specific number it just happens after some time ) not tags can be read anymore.  It always errors out after rdr.request()

System is a fresh raspian on a raspberry zero w

layereight, 2018-12-27:
Hey @zlokomatic,

can you please check whether https://github.com/ondryaso/pi-rc522/pull/49 fixes the issue for you? Thanks!

ac-astuartkregor, 2019-01-22:
@layereight I am experiencing these symptoms as well. I tried the patch (I think) but it had no effect. I can double-check soon for you.

layereight, 2019-01-23:
@ac-astuartkregor meaning https://github.com/ondryaso/pi-rc522/pull/49 fixes the issue for you now as well?

layereight, 2019-01-23:
Happy to hear that.

pitkling, 2019-10-05:
I ran into a similar problem and the patch from @layereight helped, thanks!

Interestingly enough, the problem does not occur with every RFID tag. I have two different types (tokens and cards). While the tokens work flawlessly without the patch, the cards always result in failing reads when left on the RFID reader. Once the reads start to fail, none of the cards can be read anymore **but** the tokens can be read. Once a token was read, a card can again be read again (again, resulting in errors when left on the RFID reader for too long).

Anyway, with the patch everything seems to work, thanks again!

Wyphorn, 2019-10-22:
same problem. Same fix worked. thank you!

keans, 2019-11-08:
Hm, strange with fix #49  the `wait_for_tag()` blocks only once i.e. until the first time a tag is read. Afterwards there is no waiting for tag i.e. no blocking increasing the CPU usage to 100%. 
However, activating `#self.irq.clear()` in line 399 again makes it work for me on a Pi Zero.