ondryaso/pi-rc522 #53: how to write the UID?

mrdimi, 2019-03-03:
how to rewrite the UID of the card?

MirMiz, 2019-03-13:
As far as I understood - you can't. The UID is located in the Manufacturer block (Sector 0, Block 0), which is "programmed and write protected in the production test" -> See [MIFARE datasheet, Chap. 8.6.1](https://www.nxp.com/docs/en/data-sheet/MF1S50YYX_V1.pdf) as linked in the README.md/Repo title page.
If you want to set your own RFID-Token-ID, you could write it (your own assigend Token-ID) to any block you like. For example Sector 0, Block 1. Your application then has to read it from there.

Razikus, 2020-01-09:
@MirMiz sometimes manufacturer allows to write to 0,0 in order to allow duplicating cards