ondryaso/pi-rc522 #56: stuck at wait_for_tag()

erdemontas, 2019-03-22:
I follow exactly the instructions and double check them I'am using raspberry pi 3b+
I used to work with MFRC522 library and now decided to change this one but it seems reader does not read any card at all. No warning, no error.

What should I do?

giuseppe-terrasi, 2019-04-09:
@erdemontas Did you enabled spi interface? By default, the library works on spidev0.0 so what's the output of `ls /dev/spidev*`?
I also suggest you to visit [this](https://pinout.xyz/pinout/spi) link which show where spi devices are disposed on the board.

2zimmp78, 2019-04-29:
I'm having the same problem, also on rp3B+.  I've tried all different libraries, my SPI is enabled, it's all configured up correctly... but it's not reading my tags.

giuseppe-terrasi, 2019-04-29:
@2zimmp78 Did you run the script with `sudo` or not? Also, if you are using jumper wires check that all connections are stable before run the script, in fact no error is reported if the board is not recognized. I solved the problem using an elastic:
[image]

2zimmp78, 2019-04-29:
I ran it using sudo, and my connections are secure, but it's still not reading the tags.

lnjuanj, 2019-05-21:
> I'm having the same problem, also on rp3B+. I've tried all different libraries, my SPI is enabled, it's all configured up correctly... but it's not reading my tags.

It's working for me on rp3B+ (v2.2.1). Be careful: in case the card reader is not physically working correctly, then you will not receive any information and thus it will stay at wait_for_tag forever(). In case the reader is physically ok, you will be stuck there till a RFID card is put near the reader. 
Just in case this could help.

BorisWiegand, 2020-03-15:
My SPI seems to be configured correctly on my RPi 3B+. 
Output of `ls /dev/spidev*` is "/dev/spidev0.0  /dev/spidev0.1". 

The program hangs on wait_for_tag(). How can I check if the card reader is physically working? I do not get any helpful debug message.

nifri, 2020-11-14:
@erdemontas, did you also wire up the IRQ / pin 5 on reader? Coming from a MFRC522 setup myself I hadn't wired up the interrupt and experienced the same behavior that you are describing. 

See https://github.com/ondryaso/pi-rc522/blob/master/README.md for the wiring instructions ("Connecting").

SirVer, 2021-01-05:
I seem to have the same issue. All my cables are connected correctly (according to the README) and verified with a multi meter. Still the script hangs forever at `wait_for_tag`. Do I have any means of further debugging this issue?

Twizzledrizzle, 2021-01-06:
> I seem to have the same issue. All my cables are connected correctly (according to the README) and verified with a multi meter. Still the script hangs forever at `wait_for_tag`. Do I have any means of further debugging this issue?

Did you have the IRQ connected also, that was my problem. Not sure if it was documented here

SirVer, 2021-01-07:
@Twizzledrizzle no, in fact I was incorrect about the statement that I had solid connections. I tested my cables with the multimeter and they seemed solid, but I later figured out that one of my soldered blobs was actually not connecting with the PCB at all times. Sorry for the noise and thanks for providing help and guidance!

jonnsn, 2021-09-25:
Hi,
I am having the same issue.
I have spidev 3.4 and pi-rc522 won't read any tag.
The code I used is the one provided in the usage section https://github.com/ondryaso/pi-rc522#usage.

I also tried using mfrc522 and there everything is working fine so soldering and cables should be ok (tested with multimeter anyway) and hardware itself should be too.

IRQ is connected too.

Any suggestions what might be wrong?