ondryaso/pi-rc522 #60: getting E1 or E2 error

zionnoizy, 2019-11-13:
whenever I screen the card, it only prints E1 or E2, but I have no idea where E1 or E2 comes from? I follow the instruction in here:

http://www.penguintutor.com/news/raspberrypi/rfid-rc522

how to avoid the errors to print "Detected" successfully?

nanbuwks, 2021-03-15:
In pirc522/rfid.py, lines 191 and 221, we have the corresponding code.
> print("E1")

so we have
 print("E1", file=sys.stderr)
should be

LateAtGit, 2021-03-15:
I have a similar problem, sometimes I read a tag and I get "E2" error, I tried to read the code but I have no Idea what the reason is.

Can someone help me please?

Thank you

nanbuwks, 2021-03-16:
There are several ways to solve this, but if you do not want to rewrite lines 191 and 221 of the library, you can use
You can use redirect_stdout.
In Python 3.5 or later, this can be achieved as follows:

```

import io
import os
from contextlib import redirect_stdout

from pirc522 import RFID
rdr = RFID()

f = io.StringIO()

while True:
  rdr.wait_for_tag()
  with redirect_stdout(f):
      (error, tag_type) = rdr.request()
  # print(f.getvalue())  # if need print "E1" or "E2"
  if not error:
    print("Tag detected")
    with redirect_stdout(f):
      (error, uid) = rdr.anticoll()
    # print(f.getvalue())  # if need print "E1" or "E2"
    if not error:
       print("UID: " + str(uid))

# Calls GPIO cleanup
rdr.cleanup()
```
If you do not need to capture E1 and E2, you can use

```

f = open(os.devnull,'w')
```

instead of 

```

f = io.StringIO()
```

.

LateAtGit, 2021-03-16:
Hi nanbuwks,
thank you for the answer, but I'm more interested in knowing why the error occours (and maybe solve or manage it) than suppress the message.

nanbuwks, 2021-03-19:
Since we are using radio waves, there will always be errors at the boundary of the reaction.

However, aren't you plagued by a large number of errors?

! [IMG_20200304_153805.jpg](https://qiita-image-store.s3.ap-northeast-1.amazonaws.com/0/139524/99cce50a-bc61-c658-38de-297b884aa01e. jpeg)

When I used such a module as it is, it was quite unstable as shown below.

```

Detected
Card read UID: 242,195,43,51
Selecting UID [242, 195, 43, 51, 41].
Changing used auth key to [255, 255, 255, 255, 255, 255] using method B
Calling card_auth on UID [242, 195, 43, 51, 41].
E2
Error on S1.0B0
Not calling card_auth - already authed
S1.0B0: [2] B
Calling card_auth on UID [242, 195, 43, 51, 41].
Error on S1.0B2
Changing used auth key to [255, 255, 255, 255, 255, 255] using method A
Calling card_auth on UID [242, 195, 43, 51, 41].
Not calling card_auth - already authed
Not calling card_auth - already authed
S2.0B1: []
Calling card_auth on UID [242, 195, 43, 51, 41].
Error on S0.0B0
Calling card_auth on UID [242, 195, 43, 51, 41] Error on S0.0B0
Error on S0.0B1
Calling card_auth on UID [242, 195, 43, 51, 41] Error on S0.0B1
Error on S0.0B2
.
.
.
Calling card_auth on UID [242, 195, 43, 51, 41]
Error on S15.0B2
Calling card_auth on UID [242, 195, 43, 51, 41] Error on S15.0B2
Error on S15.0B3
Changing auth key and method to None
```

To solve this problem, adding a capacitor was effective.

! [IMG_20200312_130200.jpg](https://qiita-image-store.s3.ap-northeast-1.amazonaws.com/0/139524/0f1fb004-c554-67ff-4d2c-08d9bb06b444. jpeg)

In the picture, a 47μF capacitor has been installed to improve the 3.3V power supply.