ondryaso/pi-rc522 #63: Errno 2 directory does not exist

bustercody3, 2020-02-19:
I am using a raspberry pi 4 model b running the latest version of raspbian (2020-02-13).  I followed this tutorial on how to use the rc522 RFID scanner with a raspberry pi:  https://pimylifeup.com/raspberry-pi-rfid-rc522/  /   https://www.youtube.com/watch?v=evRuZRxvPFI&t=172s  .When I got to the point where I needed to run Write.py it gave me the error:  python3:  can’t open file ‘Write.py’ :  [Errno 2]  No such file or directory

Thank you for your time, and I hope you can help.

mattelacchiato, 2020-02-20:
Are you in the same directory as Write.py? Do you have in mind that most
Linux file system are case sensitive? (Eg. Write.py is not write.py)

bustercody3, 2020-02-20:
I’m a beginner at this so I’m not entirely sure what I am doing, but here is a photo of what I have.
[image]

mattelacchiato, 2020-02-20:
The file in your folder is named "Write.py.save" .

You should clone the complete repository first and then execute the file
from within the cloned folder.
If you don't know what cloning means, you can have a look here:
https://help.github.com/en/github/creating-cloning-and-archiving-repositories/cloning-a-repository

bustercody3, 2020-02-22:
Thank you for the help, I now got past that step but now it will not recognize when I place an RFID tag on it.  The **tags came with the scanner.**  All my wires are **set up correctly** and are **soldered properly.**  When I place the tag on it it will not write **“Written”** in the terminal. Any solutions?

bustercody3, 2020-02-23:
I have now fixed it.

EnggUsama, 2020-09-22:
How you fixed it.?

bustercody3, 2020-09-24:
I fixed it by restarting the tutorial I was following from scratch, and it worked just fine the second time.

Chaoui-lpb, 2022-02-25:
Don't forget to enable SPI interface with
`sudo raspi-config`