ondryaso/pi-rc522 #67: Reader not reading certain Card

alexanderpetrenz, 2020-04-13:
Hi I bought my MFRC22 Reader on amazon and wired it up. It was shipped with a blue round dongle and with a white blank plastic card. Both are reporting to be type ISO 14443-3A MIFARE Classic 1k

Using the example program reading the blue dongle is working very stable. However it does not work at all with the white plastic card. It always returns with the error flag set to True.
I also tried reading with my android phone using the app NFC Tools. The phone reads both cards without any problem. 

I wasn't able to find any clue on how to deal with this. Are there any options I can adjust in the library regarding this issue? The cards doesn't seem to be broken, but will the behavior the same with other cards? I want to use several of the cards. Are there cards which I can safely buy or will those also be affected by this problem?

Thanks

OmegaMatze, 2020-04-22:
I have something similar.
The NXP MIFARE Classic 1k (ISO14443-3A) working fine.
But NXP MIFARE DESFire (ISO 14443-4) i don't get the full UID. they have a 7 Byte UID.
Also these are not detected by wait_for_tag.

`
rdr = RFID()
 (error, data) = rdr.request()
`

schnellerFuchs, 2022-10-22:
Hello, 
same Problem here!

If I use MFRC522 from [https://github.com/mxgxw/MFRC522-python], i can read any card.
But if I use pi-rc522 due to heavy CPU load of MFRC522, unfortunately, some of my cards and chips don't get read.

Is there a way to fix the problem?
Thanks!

schnellerFuchs, 2022-10-22:
Hello,

I have this RFID Chip: RFID Kit RC522 mit Reader, Chip und Card 13,56MHz SPI [https://www.amazon.de/AZDelivery-Reader ... B074S8MRQ7](https://www.amazon.de/AZDelivery-Reader-Arduino-Raspberry-gratis/dp/B074S8MRQ7)

(This reader is working fine on an Arduino project using RFID tags that are belonging to a key system!!)

I use this RFID chip on a python project with two libraries:

"MFRC522" https://github.com/mxgxw/MFRC522-python/
"RC-522" https://github.com/ondryaso/pi-rc522 and

MFRC522 detects all cards and other RFID tags that I use. And - if I change code a little bit - I can use 7 byte UIDs. So I could use MFRC522, but since it has a very high CPU load, I would rather use RC-522. (Yes, its known that you can reduce CPU load a little bit by adding a sleep. But it's not that good in CPU load as RC522 by ondryaso.=

So, if I use the RC522 project of ondryaso it reads only some of my tags (wired correctly). But it doesn't detect all the rfid tags I use. The code is different and something prevents it to read the key system tags...

Technically, the reader is definitely able to read all the tags. MFRC522 can read it and it works in an Arduino project.

I use it with a Raspberry Pi Model 3 B.

Thanks!

kong91896-droid, 2026-10-14:
Hi

kong91896-droid, 2026-10-14:
How are you feeling now Guy's