ondryaso/pi-rc522 #73: Reader not reading

alexanderkoller, 2021-11-07:
Hi all,

I'm trying to connect an AZ-Delivery RC522 reader to a Raspberry Pi (have tried both 3A+ and Zero 2 W), but the reader just won't read the RFID chip and card that came in the package.

I am using the [wiring with IRQ to GPIO24](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#hardware-with-fixed-pin-assignment), and have double-checked that all wires are connected correctly.

I am using the current version of Raspberry Pi OS, and have enabled SPI in raspi-config. The kernel modules are loading:
```
$ uname -a
Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux

$ lsmod|grep spi
spidev                 20480  0
spi_bcm2835            20480  0

$ ls -la /dev/spi*
crw-rw---- 1 root spi 153, 0 Nov  7 09:42 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Nov  7 09:42 /dev/spidev0.1
```

However, when I run the "util" test program from the README, it hangs on `rdr.wait_for_tag()`, no matter how long I hold the chip or card to the reader.

I have tried this both with spidev 3.5 and 3.4, as suggested in #71.

Any idea what I could try to make this work? Or how I would even debug this?

closed 2021-11-09 by alexanderkoller

Scott1803, 2021-12-19:
May I ask, if you did manage to fix this issue, how did you do it? I have come across the exact same issue you described here..

alexanderkoller, 2021-12-19:
Yes, I fixed it. In my case the problem was that I connected the RC522 to the Raspberry Pi via a GPIO extension board, and the manufacturer had accidentally connected two of the pins on the extension board. The problem went away when I connected the RC522 to the Pi directly.